hrnr/m-explore #1: indigo

KeoChi, 2016-04-12:
hello!The package can be worked on the indgo???  I have a turtlebot but I don't know how to use the package! Can you give me A detailed tutorial?

hrnr, 2016-04-12:
Hi,
are you interested in explore_lite or multirobot_map_merge package (or both)?

Although I have developed packages on jade and I'm aiming release in Kinetic Kame it might run on indigo. Have you tried building the packages?

KeoChi, 2016-04-14:
I am interested in explore_lite. And I can build the packpage sucessfully. But I launch the slam and the move_base, then I launch the  explore.launch, it log:
[ INFO] [1460604385.376109945]: Waiting for costmap to become available, topic: map
[ INFO] [1460604385.579363676, 24.590000000]: Waiting for footprint to become available, topic: move_base/global_costmap/footprint
[ INFO] [1460604386.410083224, 25.430000000]: Connected to move_base server
Why?what should I do?

hrnr, 2016-04-14:
You setup looks good. Node received map and was able to connect to move_base. I can only guess, but I think that it wasn't able to find any frontiers (exploration goals) on the map. Is your map correct?

You can get more information by turning debug logging on via rosconsole.

hrnr, 2016-04-26:
Have you been successful in setting-up the node?

KeoChi, 2016-05-03:
I have been sloving others problem. And I can do it follow the wiki :http://wiki.ros.org/explorer ! When I have time, I will try explore_lite again.

apreshant, 2016-09-26:
Hi , I'm interested in using the multi_robot_map_merge package on indigo. Can it be setup on ros indigo?

hrnr, 2016-09-27:
I do not intend to support indigo. Only jade and never will be supported. Kinetic support is on the way and will be released soon (I'm preparing new major version of `map_merge` for Kinetic)

closed 2016-09-27 by hrnr