hrnr/m-explore #2: which frontier choose?

hosh0425, 2016-08-02:
Hi,
I ran this package on ros indigo. There is more than one frontier goal and trajectory planner plans a path to one of them and robot starts moving through this path. After "planner_frequency" time outs, the frontier goals update but before arriving to the previous goal the trajectory planner choose another one. How I can optimize the frontier goals (get fewer frontier)? and what should I do in order to stop frontiers from updating before the robot has arrived to it's previous goal?
Thanks

hrnr, 2016-08-05:
Hi,

this is expected behavior. The goal is regularly updated to always head towards the "best" (largest and closest) frontier.

If you don't like this behavior, you may set planner_frequency to very small number (so that re-planning will run once per year or so). This way robot will always arrive to its original goal and only then the next best frontier will be chosen.

apreshant, 2016-09-26:
@hosh0425 : Could you please share the steps for getting this working on ros indigo.

closed 2016-09-27 by hrnr