hrnr/m-explore #3: opencv not always installed in "/usr/include/opencv-3.1.0-dev/opencv"

nburn42, 2016-09-15:
Using jade:

git clone https://github.com/hrnr/m-explore
catkin_make only found Opencv2.4 your package reqires opencv3

used this fix
sudo apt-get install ros-jade-opencv3

Now I receive this message.

CMake Error in m-explore/map_merge/CMakeLists.txt:
  Imported target "opencv_xphoto" includes non-existent path

    "/usr/include/opencv-3.1.0-dev/opencv"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in m-explore/map_merge/CMakeLists.txt:
  Imported target "opencv_xphoto" includes non-existent path

    "/usr/include/opencv-3.1.0-dev/opencv"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

nburn42, 2016-09-16:
might just be a configuration of mine, 
or might be a problem with the ros-jade-opencv3 package

apreshant, 2016-09-26:
Hi, were you able to resolve this?

hrnr, 2016-09-27:
Thanks for report. I will try with latest jade updates.

Do you have any problems with binary version or this affects just building from source?

hrnr, 2016-10-10:
Hi, the build is passing just fine on ROS build farm. I have fixed the build badge in README, just can check yourself. I 

If you have dockerfile where you can reproduce your issue, please reopen the issue. Otherwise I think  there is a problem with your installation.

closed 2016-10-10 by hrnr