hrnr/m-explore #4: suscribe to topic but does not merge

ImmoTPE, 2016-12-13:
Hello,
First thanks for this package, if I can make it work that's a great program :).
Here is the issue though, I can see my node suscribing to /big_robot/map and /big_robot/map_updates

[ INFO] [1481613140.774543296]: adding robot [/big_robot] to system
[ INFO] [1481613140.774571931]: Subscribing to MAP topic: /big_robot/map.
[ INFO] [1481613140.777035675]: Subscribing to MAP updates topic: /big_robot/map_updates.

But after that it's not merging anything, I had it to work for like 5 min it was merging one grid but now I don't have anything anymore. rostopic echo /map_merge/map gives data: [] 

My ROS version is Jade.

Do you have any idea where my problem might come from ? Thank you

hrnr, 2016-12-21:
What version are you using? Did you build from master or did you get the binary from ROS repo?

hrnr, 2017-02-10:
closing because no further info was provided.

closed 2017-02-10 by hrnr