hrnr/m-explore #5: Map is output by the merger, but it seems to just be a distorted copy of one robot

TrophyBuck, 2017-02-08:
Started two robots, both maps recognized by the merger running in "initial pose not known" mode, but the merged map that is output seems to only be a slightly distorted version of the second robot's map.

robot 1's map
[image]

robot 2's map
[image]

merged map
[image]

robot setup shown in gazebo
[image]

I've also tried setting up the robots in "initial pose known mode", though the map merger doesn't seem to recognize the ros topics I setup shown below (worth noting I did change "known_init_poses" to true when trying this).
`
rostopic pub /tbot0/map_merge/init_pose_x std_msgs/Float64 -- -0.75
rostopic pub /tbot0/map_merge/init_pose_y std_msgs/Float64 -- 0.5
rostopic pub /tbot0/map_merge/init_pose_z std_msgs/Float64 -- 0.0
rostopic pub /tbot0/map_merge/init_pose_yaw std_msgs/Float64 -- -1.5708

rostopic pub /tbot1/map_merge/init_pose_x std_msgs/Float64 -- 0.75
rostopic pub /tbot1/map_merge/init_pose_y std_msgs/Float64 -- -0.5
rostopic pub /tbot1/map_merge/init_pose_z std_msgs/Float64 -- 0.0
rostopic pub /tbot1/map_merge/init_pose_yaw std_msgs/Float64 -- 1.5708
`

hrnr, 2017-02-10:
When you see only 1 map in the output it means merger was not able to estimate tranformation between maps. The original merged needs quite a lot of overlapping area (a few rooms) to estimate transform (do the merge), maps in the pictures you show are clearly too small.

I preparing a new version with improved estimation based on my OpenCV GSOC project that you can find in the branch `new_api`. I'm planning to release it for kinetic in a few months (after OpenCV 3.2 will hit kinetic).

For `known_init_poses` mode, you need to provide initial poses as parameters (you are providing topics). It is way easier to provide parameters, you can create a launchfile for that. See [this example](https://github.com/hrnr/m-explore/blob/master/map_merge/launch/experiments/3-robots.ttt.launch).

The namespace `/tbot1/map_merge/init_pose_x` is correct assuming the map is published in `/tbot1/map`.

TrophyBuck, 2017-02-10:
Thanks for the quick response! I actually figured out the parameter issue yesterday, whoops.  Since the merge without initial positions didn't work I may try setting up a different test with more overlap for that.  In the meantime I've tried setting up the known_inital_poses mode with the following launch file, but the result is some odd behavior I've shown below.  Instead of putting the maps on top of each other properly, they've been combined by pushing the individual maps to their own corners.  I'm guessing this is an issue with what initial poses I gave- so below is the launch files for both the robots and the parameters.  The part of the robot launch file pertaining to position I've bolded.

Turtlebot Launch File
```
<launch>
  <arg name="robot_base_name"                  default="tbot"/> <!-- robot's name used by plugin to publish topics -->
  <arg name="base_prefix"                      default="mobile_base" />
  <arg name="include_fake_localization_robot0" default="true"/> <!-- assume we use simple fake localization in mutlirobot scenario for now -->
  <arg name="include_fake_localization_robot1" default="true"/>
  **<arg name="initial_pose_robot0"              default="-x -0.75 -y  0.5 - z 0.0 -Y -1.5708"/>
  <arg name="initial_pose_robot1"              default="-x  -0.75 -y -0.5 - z 0.0 -Y  1.5708"/>**

  <arg name="robot0" default="$(arg robot_base_name)0"/>
  <arg name="robot1" default="$(arg robot_base_name)1"/>

  <include file="$(find chris_turtlebot_bringup)/launch/chris_turtlebot_gazebo.launch">
      <arg name="robot_namespace"           value="$(arg robot0)" />
      <arg name="base_prefix"               value="$(arg base_prefix)" />
      <arg name="initial_pose"              value="$(arg initial_pose_robot0)" />
      <arg name="include_fake_localization" value="$(arg include_fake_localization_robot0)"/>
  </include>

  <include file="$(find chris_turtlebot_bringup)/launch/chris_turtlebot_gazebo.launch">
    <arg name="robot_namespace"           value="$(arg robot1)" />
    <arg name="base_prefix"               value="$(arg base_prefix)" />
    <arg name="initial_pose"              value="$(arg initial_pose_robot1)" />
    <arg name="include_fake_localization" value="$(arg include_fake_localization_robot1)"/>
  </include>

</launch>
```
Parameter launch file
```
<launch>
<group ns="/tbot0/map_merge">
  <param name="init_pose_x" value="-0.75"/>
  <param name="init_pose_y" value="0.5"/>
  <param name="init_pose_z" value="0.0"/>
  <param name="init_pose_yaw" value="-1.5708"/>
</group>
<group ns="/tbot1/map_merge">
  <param name="init_pose_x" value="0.75"/>
  <param name="init_pose_y" value="-0.5"/>
  <param name="init_pose_z" value="0.0"/>
  <param name="init_pose_yaw" value="1.5708"/>
</group>
</launch>
```
Result of merge with known inital pose
[image]

hrnr, 2017-02-10:
Try to set robots to face the same direction and set `yaw` to 0.0. It is a more intuitive to setup translations only.I can explain more later, I'm on phone.

TrophyBuck, 2017-02-11:
Tried that, that worked (some distortion, but I'm guessing that will get better with more overlap).  Is there a way to programmatically check if there is enough overlap for a good merge?

hrnr, 2017-02-12:
If you are using `known_init_poses` mode it does not matter how much overlap grids have. It can work even for non overlapping grids, because the transformation is known.

You can align the grids by providing more precise initial poses.

closed 2017-07-10 by hrnr