hrnr/m-explore #6: The map is not completely explored

JoBot88, 2017-06-21:
Hello,

My problem is the following, after a little while I get this message "Done exploring with 41 goals left that could not be reached.There are 44 goals on our blacklist, and 41 of the frontier goals are too close tothem to pursue. The rest had global planning fail to them" from the explore_lite node. However when I look at the map created in rviz, there is a huge part missing and as far as I can tell it would have not been so difficult to compute a path from my robot's position (at the end of the exploration) and a potentially interesting goal. Do you have any idea of what the problem could be ?

Thank you

PS: ROS version: kinetic

hrnr, 2017-07-10:
I'm aware of this bug. I will release a new version, with new algorithm for finding frontiers and better visualisation.

You can build it from source from branch [frontier_search](https://github.com/hrnr/m-explore/tree/frontier_search)

hrnr, 2017-10-31:
New version is released which fixes this bug.

closed 2017-10-31 by hrnr