hrnr/m-explore #8: Is a Indigo version will be soon available ?

StanSmooth, 2017-11-27:
Hello ! 
first of all, thank you for your work.
I would like to know if an Indigo's version of this package will be available soon ?

Thank you !

hrnr, 2017-11-27:
Hi,

currently supported are only Kinetic (LTS) and Lunar. I don't plan to support or release packages for Indigo. I suggest to upgrade to Kinetic or Lunar.

If you need to run on Indigo, you need to build it yourself, but beware, there might be some patching needed.

StanSmooth, 2017-11-27:
thank you for your answer. 
Unfortunately I don't have the skills required to do it !

hrnr, 2017-11-27:
Maybe it is not as hard as it sounds. Building catkin packages is not too difficult (usually just just need to invoke `catkin_make`) and there are some tutorials:

* http://wiki.ros.org/catkin/Tutorials/using_a_workspace
* http://wiki.ros.org/catkin/Tutorials

Upgrading ROS requires usually 3 steps and I'm not aware of any tutorial on that:

1. remove all ROS packages, backup installed package names
2. upgrade Ubuntu
3. install all your previous ROS packages for the new distro 

You can find a lot of tutorials on upgrading ubuntu.

I would recommend to upgrade your ubuntu to Xenial LTS and use ROS Kinetic. Then you can easily install released packages for ROS Kinetic.

hrnr, 2017-11-27:
duplicate of #1.

closed 2017-11-27 by hrnr