hrnr/m-explore #9: Costmap is not generated properly

egemenertugrul, 2017-11-27:
The thing is, I don't have a real turtle and I'm publishing the TF and Scan data from Unity3D. The connection is fine. I now use your explore.launch to determine the locations to explore with "rosrun move_base move_base" running in another terminal. The problem is that the costmap is not set properly. 

[image]

As you can see, the costmap (visualized like a map) is not well set up.

First, I launch the `explore.launch` in a terminal. When it awaits for move_base node, I run `rosrun move_base move_base` on another terminal. The terminal outputs are as follows:

roslaunch explore.launch: https://gist.github.com/egemenertugrul/92fc3f2ec23efc5c789d16af23ccf383
rosrun move_base move_base: https://gist.github.com/egemenertugrul/c8224e5653dc23acb0da98a22c89be4c

I'm kinda lost to solve this problem. Only if I could set my costmap properly, I'm guessing that the move_base would work properly. 

I don't need to publish Twist command from move_base. I only need a pathfinding algorithm to find the path to the goal published by your explore.cpp, outputting a Pose array. I will be sending this array back to Unity.

Any ideas?

hrnr, 2017-11-27:
`move_base` is one of the packages that is quite hard to setup. It needs a lot of parameters in different namespaces. Runnning just `rosrun move_base move_base` is almost guaranteed *not* to work. You need to setup `move_base` correctly before experimenting with `explore_lite`. See this tutorial http://wiki.ros.org/navigation/Tutorials/RobotSetup

When you are able to control your robot from rviz via "2D nav goal", you can start using `explore_lite`. 

The only specific setup for `explore_lite` you need: You should setup global planner to allow planning through the unknown space, but that should be the default now. Described here http://wiki.ros.org/global_planner#Parameters

I will clarify this on the wiki page.

egemenertugrul, 2017-11-27:
After I properly set up move_base (as shown in [http://wiki.ros.org/navigation/Tutorials/RobotSetup](http://wiki.ros.org/navigation/Tutorials/RobotSetup)), the costmap was generated and explore_lite worked flawlessly! 

[image]

The only difference for me was to exclude the TF, Sensor and Odometry publishers (because I'm already publishing them over Unity3D). It was a bit lazy for me not to check how move_base should've been set up in the first place. Thank you so much for the support and such a useful asset.

closed 2017-11-27 by egemenertugrul