hrnr/m-explore #11: Node explore cannot be found after launching explore.launch

kenanEkici, 2017-12-10:
This is what I get when I try to run explore.launch in the explore_lite package.

ERROR: cannot launch node of type [explore_lite/explore]: can't locate node [explore] in package [explore_lite]

What am I doing wrong?

hrnr, 2017-12-11:
This probably means you don't have this package installed correctly.

To resolve this I need more information.
* How did you installed explore_lite?
* package version
* which ros distribution you are using

It might be also problem with your specific installation of ROS. Do other packages work correctly?

kenanEkici, 2017-12-13:
I have installed explore_lite with: 
sudo apt install ros-kinetic-multirobot-map-merge ros-kinetic-explore-lite
I'm not sure which package version I have cloned the master
And I'm using Kinetic.

I'm not entirely sure what you mean with other packages?

hrnr, 2017-12-13:
I don't understand. Have you installed your package from repository, or have you cloned from master and compiled it yourself?

Does other packages works as expected? For example for explore_lite you need to have configured `move_base`. Does `move_base` work? (Can you navigate your robot with `move_base`?)

kenanEkici, 2017-12-13:
We are trying to configure it the way it's mentioned in the quick-start.txt

hrnr, 2017-12-13:
That file is outdated. I should probably remove that.

kenanEkici, 2017-12-13:
Can you provide me a recent quickstart for ROS Kinetic for Turtlebot2

hrnr, 2017-12-13:
I have removed the mentioned file. Thanks for pointing that out. I have forgotten about that file.

I think there is no quickstart no longer needed. Package is now properly released in ROS, so installation is very easy. `quickstart.txt` was just about installation of early development version of this package.

I can reproduce the error you have. Thanks for reporting. I will fix it soon.

hrnr, 2017-12-13:
As a workaround you can uninstall packages via apt and clone this repo to your workspace. From workspace everything works as expected.

kenanEkici, 2017-12-13:
What I did was: 
- I uninstalled the packages
- removed the repo from source 
- cleaned the build
- recloned 
- catkin_make 

And from here we are stuck because we are not sure how to launch the package because nothing seems to be working

kenanEkici, 2017-12-13:
Okay we have successfully been able to run explore_lite explore.launch, after doing a minimal launch on the turtlebot. What are the next steps to make the robot running autonomous? Thank you in advance

hrnr, 2017-12-13:
you need to have correctly configured `move_base`. You should be able to navigate the robot with move_base manually. Then explore.launch should work for your more or less out of the box (I was also using that launch file with turtlebot).

Of course, you might need to change some topic names and tf frames names depending on your configuration. Check the wiki how to configure the parameters.

kenanEkici, 2017-12-13:
Okay, we will try that. Do you perhaps have the launchfile for a Turtlebot?

hrnr, 2017-12-13:
the launchfile shipped with explore_lite is the same launchfile that I have used for experiments.

kenanEkici, 2017-12-13:
Okay, we have an issue with configuring the move_base. Our move_base package is empty except for a package.xml file.

hrnr, 2017-12-13:
ok, come back when you get your `move_base` up and running and let me if you have any further issues with explore_lite.

Thanks again for reporting this issue, I will fix it in the next package release, so the package from ROS repository works out of the box.

kenanEkici, 2017-12-14:
After configuring move_base, we have been able to successfully launch explore.launch. It works very well, but some parameters need to be tweaked in order to make it run more efficient.

hrnr, 2017-12-14:
Great. Unfortunately filtering of small frontiers is now hard-coded because of #7 (but it is fixed in master, so it should not affect you). You can still tune weights of frontiers (potential_scale, gain_scale), which affects the exploration considerably.

If you have some other issues or suggestions for improvements let me know.

kenanEkici, 2017-12-14:
Thanks. The only issue we encountered since then is mapping small and crowded rooms (with or without moving obstacles). It constantly keeps trying to recover and eventually never accomplishes to map the room and therefor never moves out of a specific room (atleast the smaller ones). If finetuning it doesnt't help we will be considering to try and fix this.

hrnr, 2017-12-14:
Oh yeah. That is a general problem with `move_base`. Exploring can't do much about it. Maybe exploring should avoid such spaces, but I'm not sure if that is viable.

Tweaking parameters of move_base usually helps .I had some success with big inflation on global map (to keep it from walls) an small inflation in local map (to leave the planner as much space as possible).

commit c3ac7b14 (2017-12-15): fix bugs in CMakeLists.txt

* install nodes in packages, so they get shipped in debian packages. fixes #11
* add dependencies to exported targets for all targets. This is required by catkin. Probably wasn't causing any troubles, since we use only precompiled messaged shipped with ROS.
* add necessary dependencies to catkin_package macros (mostly messages we depend on)
* fix issues reported by catkin_lint
	* sort all lists
	* minor warnings
closed 2017-12-15 by hrnr