hrnr/m-explore #12: Suscribes map topic but does not merge, Unable to compute transforms

dharmp15, 2018-04-18:
Hi,
I am using kinetic version of this package. It is discovering all robots and map topics perfectly but not able to merge the maps. the empty (no data) merged map is coming at the rate approx 4. 
I have tried in both cases, known and unknown initial poses but it seems map_merge node is unable to compute the transforms.
In known case, for each robot it is getting initial poses but transform looks only NAN values.

Thanks for any help!...

hrnr, 2018-04-18:
How do the maps you are trying to merge look?

Have you tried your setup with maps that are used in tests (just publish them with map_server)? see https://github.com/hrnr/m-explore-extra and https://github.com/hrnr/m-explore/blob/717e7f6ea7dfc42cc5a403fff5088f2d50602c6c/map_merge/test/test_merging_pipeline.cpp#L95

dharmp15, 2018-04-19:
Hi hrnr,
Thanks for prompt reply. Now merging  problem is solved. It is merging perfectly in both the cases. I have installed the package from my local Repo and it was old version, just replaced with new master version.
But now I am facing new problem that the merged map is coming with some offset (may be you call it origin of merged map). I am attaching picture of maps for more detail. what could be the possible reasons? and any suggestion. I am using octomap_server to create occupancygrid map.
[image]

--

MaxGsomGsom, 2018-04-19:
I think it is bug, because  map merger does not take into account offset of original maps. You can try this  workaround:
Set ``world_frame`` param to ``map_merged``
Add this to your launch file:
```
<node pkg="tf" type="static_transform_publisher" name="map_merged_tf" args="-halfX -halfY 0 0 0 0 world $(arg r_ns)$(arg r_id)/map_merged 100" />
```
where halfX, halfY - a half of size of the map

dharmp15, 2018-04-19:
thanks @MaxGsomGsom ,
I am trying online map merging with increasing size of maps and the initial size of robot's individual  maps depends upon environment around the robots (halfX, halfY will be random and unknown). So it will not solve my problem. And initial offset of robot frames I am providing through static tf.

hrnr, 2018-04-20:
The merged map is in the world tf_frame. Sadly map_merger is not providing any tf transform (world -> robotX). It is on my TODO list, but it has never been implemented as I didn't need it for my research. 

If you are providing those transforms yourself, you need to adjust them, so that the merged map is aligned with the robots.

Another thing is that the merged map has its origin in the (0,0) corner, so it shows off the centre in rviz (in the world frame). If you agree I can put the origin of the map in the centre of the grid, if that seems more natural. That might break setup for some people, but will be probably ok for release to the new ROS.

dharmp15, 2018-04-21:
"If you agree I can put the origin of the map in the centre of the grid, if that seems more natural. " I thing this would solve my problem because maps always builds around the robots. Tell me how can I do in code. I have all tfs available with me.

hrnr, 2018-04-21:
Implemented in d59acfab620faacbbaa183b0fb6e5f518894dc70

Merged grid has its origin set to its centre. You still need to provide correct tf transforms by yourself, but as you say that should no be an issue for you.

Note that still the origin might be in an inaccessible area.

dharmp15, 2018-04-22:
1. We both are thinking totally 180 deg apart.
yours initial poses====>Tfs 
and mines Tfs=====> initial poses
two points I would like to mention here:

** plan to implement publishing of tf transformations world -> robotX by map_merger====>This will always create conflict as SLAM algorithms are already publishing the same tfs. And let SLAM algorithms take care about TFs as map merger will also take data from SLAM only to publish tfs.
** When robots are starting from different positions it is usually very hard to measure initial positions:==> in out door case (GPS available area) we can easily find out initial poses of robots form its TFs/odometry. And in case of any failure of localization node like, some random behaviour of robot, power failure, loose connections etc when we restart it will take current position as its initial position and no need to bring robot at initial position or measuring again position of robot in world frame.

dharmp15, 2018-04-22:
Yes. In general GPS does not provide such high Accuracy that we can use it for merging. Just try to find out how to remove these offsets and bring map and robots in correct position, then merged map can be reused for online multi robot exploration and mapping.
Thanks.

closed 2018-04-22 by dharmp15