hrnr/m-explore #13: map_merger is subscribing to two map topics, but only displaying one

rkleinma, 2018-04-19:
I took a look at the other post on this type of issue, but it does not seem to apply in this case. I pulled the latest version of the Master, but the issue still remains.

I am trying to run map_merger from two bag files using two namespaces, let's say robot1 and robot2. Each bag file (robot1.bag and robot2.bag) contains a map topic under their unique namespace (i.e. /robot1/map). This is our process:
1) run roscore
2) set use_sim_time true
3) run the robot1 bag file with clock "rosbag play robot1.bag --clock"
4) run the robot2 bag file "rosbag play robot2.bag"
5) run map_merger "roslaunch multirobot_map_merge map_merge.launch"

When viewing in rviz, the output from map_merger (map_merge/map) appears to be a copy of robot1/map (updating at a slower rate), but does not merge any part of robot2/map. However, map_merger subscribes to both and prints to the terminal that it has identified matches (~15-20).

I'm also having trouble running the provided test files. Is there a document outlining how to run these correctly? Maybe that would help our issue?

hrnr, 2018-04-20:
The tests can be run in the standard way 

```
catkin_make run_tests
```

The test data should be automatically downloaded for you. These tests are checked by ROS CI. See for example

http://build.ros.org/job/Kdev__m_explore__ubuntu_xenial_amd64/lastBuild/testReport/(root)/MergingPipeline/

If you think your maps have enough overlapping space and therefore should have been merged, I need those maps in format for map_server (.pgm + .yaml) to be able to reproduce the issue (you can save them with map_server).

hrnr, 2018-04-25:
Thanks for reporting you helped to fix a bug (06de4b6).

TL;DR: You need to improve your mapping accuracy and have more overlapping space.

Problem is that that your maps are really small and that your mapping is not very precise. You can see they are bended in opposite ways. Therefore the match between your maps has very low confidence (rightly), you need to decrease confidence parameter to see the match (use latest master with 06de4b6). You can use `from_map_server.launch` to quickly evaluate your maps.

If you compile the merger in debug mode you can some more info.

```
0 1
features: 85 55
matches: 13
inliers: 4
inliers/matches ratio: 0.307692
confidence: 0.336134
[-1.117468898188666, -0.09726900163759883, 878.5337524365971;
 0.09726900163759883, -1.117468898188666, 417.2427239080802;
 0, 0, 1]
```

So the match relies only on 4 points. There is a low change that this will be a correct match (that is why it is rejected by default). As expected the result is not correct

[image]

Debug mode also produces matches and inliers visualisation to see why

[image]
[image]

closed 2018-05-04 by hrnr