hrnr/m-explore #14: Bug in merging_pipeline

MaxGsomGsom, 2018-04-19:
Crash when ``grids_[0] = null``
https://github.com/hrnr/m-explore/blob/717e7f6ea7dfc42cc5a403fff5088f2d50602c6c/map_merge/src/combine_grids/merging_pipeline.cpp#L158-L159
Possible fix
```
  for (auto grid: grids_)
    if (grid) {
        result->info.resolution = grid->info.resolution;
        break;
    }
```

hrnr, 2018-04-20:
Thanks for reporting this. Actually we need to find a grid, which is a reference frame, in case we have multiple resolutions.

hrnr, 2018-04-21:
Thanks again for the report. I have fixed this along with some other issues. The crash is fixed and the behaviour when there is no merge possible should be improved.

commit 5ab98aa9 (2018-04-21): fix nullptr derefence while setting resolution of output grid

* fixes #14
* when one of the maps didn't arrived yet, nullptr derefence might happen
* set the correct resolution in case we are working with multiple resolutions (according to the selected reference frame)
* improve behavior when we can't merge anything. Now we try even harder to output at least non-empty map.
* add new test covering covering the bug. improved existing tests.
closed 2018-04-21 by hrnr