hrnr/m-explore #15: The second robot map seems not in the right position

Vincent1923, 2018-05-04:
Thank you for the package. It is discovering all robots and map topics perfectly. But there was some problem when it merged the maps. The two maps were published through two rosbag. When I used the “known_init_poses” mode, it seems the transform between the two maps is not right. I call the two maps “/robot1/map” and “/robot2/map”. And I set the world_frame to “/robot2/map”. The maps published by map_server are as follows. The starting point of both maps in the ground is the same.

map1.pgm:
[image]
map1.yaml:
[image]

map2.pgm:
[image]
map2.yaml:
[image]

Here is the map_merge.launch file and I add the initial poses in it.
`<launch>
<group ns="map_merge">
  <node pkg="multirobot_map_merge" type="map_merge" respawn="false" name="map_merge" output="screen">
    <param name="robot_map_topic" value="map"/>
    <param name="robot_namespace" value=""/>
    <param name="merged_map_topic" value="map"/>
    <param name="world_frame" value="/robot2/map"/>
    <param name="known_init_poses" value="true"/>
    <param name="merging_rate" value="0.5"/>
    <param name="discovery_rate" value="0.05"/>
    <param name="estimation_rate" value="0.1"/>
    <param name="estimation_confidence" value="1.0"/>
  </node>
</group>

<group ns="/robot1/map_merge">
  <param name="init_pose_x" value="-122.356226"/>
  <param name="init_pose_y" value="-97.080847"/>
  <param name="init_pose_z" value="0.0"/>
  <param name="init_pose_yaw" value="0.0"/>
</group>
<group ns="/robot2/map_merge">
  <param name="init_pose_x" value="0.0"/>
  <param name="init_pose_y" value="0.0"/>
  <param name="init_pose_z" value="0.0"/>
  <param name="init_pose_yaw" value="0.0"/>
</group>

</launch>`

Here is the merged map and I set the world_frame to “/robot2/map”. But the “/robot1/map” is not in the right position. 
[image]

I don't know if the parameters I set are correct. Could you please give me some help? Thank you!

Vincent1923, 2018-05-04:
Thank you for your prompt reply!
I have tried merging the maps without known positions. I could not get the merged map as the two maps do not have much overlapping space.
[image]

I will appreciate you for helping me have a look at my maps. [Here is my original .pgm + .yaml files for both maps.](https://drive.google.com/open?id=1Y9ea_ARx8OV_PRSx5QbfnQ1QmZVj9wUp)

Thank you very much!

hrnr, 2018-05-04:
1. Your maps seems to be broken. Marking of the free space, unknown space and occupation values are switched. What is the source of your maps? The maps should have correctly marked free space and unknown space according to the specification of the of the occupancy grid msg http://docs.ros.org/api/nav_msgs/html/msg/OccupancyGrid.html

2. Are we using the same maps? I'm getting quite different results.

```
0 1
features: 39981 28580
matches: 628
inliers: 3
inliers/matches ratio: 0.00477707
confidence: 0.0152749
[-0.07158130290853876, 0.0264739041042349, 2523.265982276639;
 -0.0264739041042349, -0.07158130290853876, 2424.609337158685;
 0, 0, 1]
```

3. What is the expected transformation between maps origins?

Vincent1923, 2018-05-06:
Thank you for your answer. I am sorry that I reply so late!

1. I use the google's cartographer_ros to get the maps. And I used the map_server to save the two maps. But I have changed a little of the source code of the “map_saver.cpp” in it. I think this is the reason why the maps looks different. 

2. We are using the same maps, but I got the results above when the two maps was generating.

3. I want to merge the two maps but I don’t know if the parameters I set are correct.

hrnr, 2018-05-07:
1. You need to fix the marking of free space, unknown space and occupation. Even with correct transformation your maps can not be merged. Think about overlaying the values.

2. makes sense

3. when you use merging with known initial positions you are responsible for entering the correct transformation (initial position). Therefore the parameters (your initial position) were not correct.

hrnr, 2018-05-07:
Also, when saving maps make sure to select lossless format. Your pngs are full of compression artefacts.

Vincent1923, 2018-05-07:
1. Dose it means that if I don't fix the marking of free space, unknown space and occupation in my maps, I can't get the correct merged map even with setting the correct parameters?

2. Dose the correct transformation (initial position) means the lower left corner of each map?

3. The map format I use is pmg. Do you mean my pgms are full of compression artefacts?

4. If the yaml files of the two maps are as follows, could you please tell me the correct parameters?

map1.pgm:
[image]
map2.pgm:
[image]

hrnr, 2018-05-07:
1. yes

2. it means the true starting possition of the robot in your frame of reference

3. oh sorry, that was just the interpolation of my viewer. pgm is fine, always lossless.

4. you can not tell that from the yaml files. You need to know true starting positions of the robots. For example measure the distance in x,y,z and angular difference between 2 robots in meters (with a ruler or your own localisation system).

Vincent1923, 2018-05-08:
1. The true starting positions of the robots are the same in the ground, and it is the position I have pointed out in the first picture and the second picture above. 

2. I have changed the marking of free space, unknown space and occupation in my maps. [Here are the original .pgm + .yaml files of the new maps.](https://drive.google.com/open?id=11b9qgzoRs3bq0CnctrtdXSQXYwlaLEtS) How can I set the parameters to get the correct merged map? 

Thank you!

hrnr, 2018-05-17:
the problem with maps seems to be fixed. I'll take a look, but it'll take while since I'm busy these days.

closed 2018-09-04 by Vincent1923

MerryMarry, 2018-12-07:
Hello
if you figured it out pls tell us
thank you

Sadighi-Nicky, 2018-12-08:
If you are using gmapping, try to set the zero point in the middle of the map like this in your launch file:

    <param name="xmin" value="-200" />
    <param name="xmax" value="200" />
    <param name="ymax" value="200" />
    <param name="ymin" value="-200" />

then set the initial positions in your merging launch file to 0.0. This works fine for me, hope it helps you too.

MerryMarry, 2018-12-08:
> If you are using gmapping, try to set the zero point in the middle of the map like this in your launch file:
>
> ```
> <param name="xmin" value="-200" />
> <param name="xmax" value="200" />
> <param [...]

Thanks