hrnr/m-explore #16: When giving known positions assertion appears

ccimrie, 2018-06-11:
Hi,

If I provide the following the launch file
``` xml
<launch>
  <group ns="/robot1/map_merge">
    <param name="init_pose_x" value="0.068853"/>
    <param name="init_pose_y" value="-0.708363"/>
    <param name="init_pose_z" value="0.035003"/>
    <param name="init_pose_yaw" value="-0.031117"/>
  </group>
  <group ns="/robot2/map_merge">
    <param name="init_pose_x" value="-0.206623"/>
    <param name="init_pose_y" value="1.092293"/>
    <param name="init_pose_z" value="0.034994"/>
    <param name="init_pose_yaw" value="-0.340292"/>
  </group>
    <group ns="map_merge">
      <node pkg="multirobot_map_merge" type="map_merge" respawn="false" name="map_merge" output="screen">
        <param name="robot_map_topic" value="map"/>
        <param name="robot_namespace" value=""/>
        <param name="merged_map_topic" value="map"/>
        <param name="world_frame" value="map"/>
        <param name="known_init_poses" value="true"/>
        <param name="merging_rate" value="0.5"/>
        <param name="discovery_rate" value="0.05"/>
        <param name="estimation_rate" value="0.1"/>
        <param name="estimation_confidence" value="1.0"/>
      </node>
    </group>
</launch>
```

and set known_init_poses to false then it will work fine, but if I set it to true then I get the following assertion error:
[image]

And I'm not sure what this means.

hrnr, 2018-06-12:
Thanks for reporting. The bug is fixed in 745ae1e79a8289ca825cce7f7382e3e611073171

Use current master to receive the fix.

ccimrie, 2018-06-12:
Hi,

Thanks for getting back so quickly!

That's it working, though the merged map produced is a little off. I have 4 robots in a room in Gazebo (they're the 4 black dots in the bottom left room):

[image]

I'm only using concerned with merging three of the robots' maps, and below are there produced maps:

[image]
[image]
[image]

Using the XML file:
``` xml
<launch>
  <group ns="/robot0/map_merge">
    <param name="init_pose_x" value="40.003349"/>
    <param name="init_pose_y" value="-20.011330"/>
    <param name="init_pose_z" value="0.180003"/>
    <param name="init_pose_yaw" value="-0.000910"/>
  </group>

  <group ns="/robot1/map_merge">
    <param name="init_pose_x" value="39.964791"/>
    <param name="init_pose_y" value="-22.008312"/>
    <param name="init_pose_z" value="0.180003"/>
    <param name="init_pose_yaw" value="-3.045119"/>
  </group>

  <group ns="/robot2/map_merge">
    <param name="init_pose_x" value="40.005176"/>
    <param name="init_pose_y" value="-22.017528"/>
    <param name="init_pose_z" value="0.180003"/>
    <param name="init_pose_yaw" value="-0.001420"/>
  </group>

    <group ns="map_merge">
      <node pkg="multirobot_map_merge" type="map_merge" respawn="false" name="map_merge" output="screen">
        <param name="robot_map_topic" value="map"/>
        <param name="robot_namespace" value=""/>
        <param name="merged_map_topic" value="map"/>
        <param name="world_frame" value="map"/>
        <param name="known_init_poses" value="true"/>
        <param name="merging_rate" value="0.5"/>
        <param name="discovery_rate" value="0.05"/>
        <param name="estimation_rate" value="0.1"/>
        <param name="estimation_confidence" value="1.0"/>
      </node>
    </group>
</launch>
```
however produces this output:

[image]

Where two of the robots are merged together at the opposite side of the other robot?

(Sorry for potentially too many pictures!)

hrnr, 2018-06-13:
The initial positions you are giving seems weird. Are you sure they are correct? The merged map clearly contains one map rotated. That would be the map for robot1, since you have put rotation to its `init_pose_yaw`.

Why the headings are not the same? To me the roots seems to be aligned with the coordinate system and they seems to have the same heading.

If you are sure the initial positions are indeed correct, there might be a problem with `map_merge` handling of map origins. You might need to offset the origins manually.

I still need to investigate the issue with map origins, see #15. It would help if you could provide the maps in .yaml+.pgm `map_server` format.

As a workaround just adjust the initial positions you are giving until you get the correct merge. If you let me know what transformation was needed to initial positions to get the correct merge, I can then fix map_merge to do it automatically. I believe we just need to transform the initial positions by the map origins, but I might be wrong.

closed 2018-06-16 by hrnr