hrnr/m-explore #17: Release for Melodic

hrnr, 2018-06-12:


hrnr, 2018-08-18:
waiting for `move_base` to become available in Melodic

hrnr, 2019-05-13:
`move_base` is in. there are no blockers right now. pending free time.

rmerriam, 2020-02-23:
I don't find the Melodic version in the ROS system. I cloned and compiled and it works. Will a release be made?

rojinphilip, 2020-09-01:
> I don't find the Melodic version in the ROS system. I cloned and compiled and it works. Will a release be made?

Hi, How did you compile it? I'm trying it use it for turtle bot [tutorial. ](https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#virtual-slam-by-multiple-turtlebot3s)

Thank you.

Timple, 2020-09-25:
@hrnr If free time is still an issue, I can do the release if you like.

hrnr, 2021-01-03:
Release for Melodic is in PR: https://github.com/ros/rosdistro/pull/27878

closed 2021-01-03 by hrnr