hrnr/m-explore #19: Problem with map-merging without initial pose.

hzhang80, 2018-08-27:
Hi,
    Thanks for the package, it's very helpful but I met some problems when running some tests.
    I used two turtlebot3 burgers in gazebo to do some sim tests. When the package knew their initial pose, it worked very well, but when I set the parameter to False, the world map provided by the package became to only the local map of robot1.
    I thought maybe it needed more overlap to calculate the transformation, so I ran the simulation for another several times. But even when the map of robot1 was included in the map of robot2 totally, the world map(i.e. the merged map) was still same as the map of robot1.
    This was strange because according to the log info, everything worked fine and maps were "merged successfully".
     Do you have any suggestions? Thank you very much, looking forward to your reply.

Best,
Hui.

hrnr, 2018-08-28:
Hi Hui,

to get more info about the map-merging process you can rebuild the package in DEBUG mode, it produces more verbose output. From your description, it seems that the merger was unable to estimate the transformation between maps.

Feel free to post the log (relevant parts) and maps from map_saver (yaml + pgm), if you think there is enough overlapping space.

Cheers,
Jiri

hzhang80, 2018-08-28:
Thanks for the prompt reply! Following are my simulation environment, log info, and maps of robot1, robot2 and merged map respectively. GitHub doesn't support yaml and pgm files so I'll post screencuts and text, sorry for that.
I've read your paper for this package, it's so great. May I ask that where can I get the log info includes the transform matrix?
Thank you again.
[image]
[DEBUG] [1535479710.250528972, 1109.967000000]: Map merging started.
[DEBUG] [1535479719.213659145, 1112.146000000]: received full map update
[DEBUG] [1535479726.980209264, 1114.101000000]: pairwise matching features
[DEBUG] [1535479727.063435150, 1114.123000000]: calculating transforms in global reference frame
[DEBUG] [1535479727.063627869, 1114.123000000]: optimizing global transforms
[DEBUG] [1535479727.063824668, 1114.125000000]: Grid pose estimation started.
[DEBUG] [1535479727.063954300, 1114.125000000]: computing features
[DEBUG] [1535479739.007220216, 1117.573000000]: received full map update
[DEBUG] [1535479742.074680197, 1118.361000000]: received full map update
[DEBUG] [1535479744.128180174, 1118.947000000]: pairwise matching features
[DEBUG] [1535479744.218285255, 1118.978000000]: calculating transforms in global reference frame
[DEBUG] [1535479744.218622716, 1118.978000000]: optimizing global transforms
[DEBUG] [1535479744.218843779, 1118.978000000]: warping grids
[DEBUG] [1535479744.240294263, 1118.983000000]: Grid pose estimation started.
[DEBUG] [1535479744.340818470, 1119.020000000]: compositing result grid
[DEBUG] [1535479744.358216577, 1119.027000000]: computing features
[DEBUG] [1535479744.369632329, 1119.030000000]: all maps merged, publishing
[image]
image: robot1.pgm
resolution: 0.050000
origin: [-100.000000, -100.000000, 0.000000]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.196
[image]
image: robot2.pgm
resolution: 0.050000
origin: [-100.000000, -100.000000, 0.000000]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.196
[image]
image: total.pgm
resolution: 0.050000
origin: [0.000000, 0.000000, -nan]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.196

hrnr, 2018-08-29:
> GitHub doesn't support yaml and pgm files so I'll post screencuts and text, sorry for that.

could you use some other service and post a link? I can't reproduce the issue with screenshots. From your screenshots, it seems there are quite some mapping errors; walls are not straight and some angles are not correct. In simulator SLAMs often have issues, because there is not enough landmarks. Maybe this leads map-merging to fail?

> May I ask that where can I get the log info includes the transform matrix? 

recompile the package in debug mode. it will print more info.

Unfortunately the logs you have posted are not helpful in determining why the maps cannot be merged. Try recompiling in debug mode, it should output more information.

hzhang80, 2018-08-29:
Here are the maps.
[https://drive.google.com/open?id=1siRBZ5IIALPeggQ5gyUve2cxuVurYgNu](url)

I used gmapping to do the mapping work, in my tests the maps generated by it always have some drift(in the simulator or in the real world). Maybe it's the reason, I will tune the parameters to see if I can make the map better.

Will recompile the package in debug mode, and I'll post what I get.

hzhang80, 2018-08-29:
I ran this test in a new maze which contains several trash cans as landmarks to decrease the drifts, this time the merging was successful. The maps are in link below. 
https://drive.google.com/open?id=1fahzNYrWRhyqN3cznX5mAbrdBfH4-H81

though the merging was successful, the overlap it needed seemed to be too much. Each of the robots explored almost the whole maze when the maps were merged. Do you have any advise for this?

hrnr, 2018-08-30:
I have run your maps, which you have kindly provided. For the first pair I got this additional info in the debug mode:

```
[DEBUG] [1535636337.758969281]: pairwise matching features
0 1
features: 471 441
matches: 94
inliers: 21
inliers/matches ratio: 0.223404
confidence: 0.58011
[0.9805939018244119, 0.0149403109787057, 5.368679216168641;
 -0.0149403109787057, 0.9805939018244119, 70.11126975140731;
 0, 0, 1]
[DEBUG] [1535636338.785680746]: warping grids
```
Note that the merging algorithm was able to estimate the transformation with 21 inliers, the confidence is 0.58011. If you want to use this transformation you need to set `estimation_confidence` param lower that the estimated confidence. With `estimation_confidence=0.5` we get the merged map:

[image]

Double walls are caused by mapping errors. Generally environments like yours with low number of features will cause problems to the both mapping and map-merging. That's why the map-merging has quite a low confidence in the estimated transform. This is a rare case in robotics, where you will usually get better results in the real world than in the simulation.

For the second pair of maps there are more inliers:

```
0 1
features: 480 600
matches: 131
inliers: 48
inliers/matches ratio: 0.366412
confidence: 1.0148
[0.9973614068824546, 0.0522918986581147, -105.4672661106791;
 -0.0522918986581147, 0.9973614068824546, 109.0125554032877;
 0, 0, 1]
```

If you want to get the merged map early (with less overlapping space) I think lowering the `estimation_confidence` param should help here too.

hzhang80, 2018-08-30:
Thank you. I think my question has been solved.

hzhang80, 2018-08-30:
sorry I still have a question, could you tell me how to recompile the package in debug mode please? the package is not installed in catkin_ws but in the opt folder, I'm not sure if I can just do the "catkin_make -DCMAKE_BUILD_TYPE=Debug" in there

hrnr, 2018-08-31:
You need to remove the binary package, put sources into your catkin_ws and rebuild with `catkin_make -DCMAKE_BUILD_TYPE=Debug`. I think I will reconsider this to make the extended logging switchable on runtime.

There has been a bug affecting setting `estimation_confidence`. https://github.com/hrnr/m-explore/commit/06de4b609d002ca3111e7226ddb7fa08e058b202 I need to release a new version, as soon as ROS Melodic gets `move_base` I will release.

hzhang80, 2018-08-31:
I've changed it to debug mode and I'm using your source codes. It works well, thank you.

jfeldman24, 2019-04-16:
I am currently having a similar problem. You can see in the below image I have two robots sitting in nearly the same spot (shown as the two bottom maps). However, the merged map is only using the data from the left map. I understand this particular example has few features, but even by reducing the confidence down to 0.1 is not causing the maps to be merged. I have attempted with much more complete, feature rich maps than this as well to no avail. Any suggestions?

[image]

[image]

hrnr, 2019-05-13:
@jfeldman24 please open a new issue if you still have some questions. I'm closing this one as the original question/problem was solved.

closed 2019-05-13 by hrnr