hrnr/m-explore #21: Save map after merging

tranducthuan1, 2018-10-29:
Hi everyone,

I could run well the code for merging the map.

However, I could not save the map after merging.

Could you tell me how to save the merging-map, please.

Yours sincerely,
Tran

hrnr, 2018-10-29:
use [map_saver](http://wiki.ros.org/map_server#map_saver)

closed 2018-10-29 by hrnr

tranducthuan1, 2018-10-30:
@hrnr , thank you so much for your comment.

However, I could not save map using the map_saver.
after I run " rosrun map_server map_saver -f tran" it shows "waiting for the map.
using "rostopic list", The output topics consists of /map_merge/map, robot1/map, robot2/map.

Do we have to wait for the merge-map program finish, then save map.

I run a tutorial example, but the program continues to run after 12 hours, so I canceled it.

Could you give me the advice to solve it, please

hrnr, 2018-10-30:
map_saver by default uses /map topic. You need to remap that to the topic
you want to save. See https://github.com/ros-planning/navigation/issues/253

tranducthuan1, 2018-10-31:
@hrnr ,

Thank you very much, I could save map using remap.

Yours sincerely,
Tran