hrnr/m-explore #22: Map merging without initial pose only publishing one of the maps

MerryMarry, 2018-11-08:
Hello
We are having a group project in ROS using P3AT robots and Gmapping.
We wanted to use your package to merge the maps published by each of the robots but we have faced some problems.
I have read the previous issue on this project where you changed the estimation confidence to merge the maps properly but no matter how much I change it, the only thing that happens it that the merged map changes from robot0 to robot1's map and keeps switching between them, so that we can't manage to get the merged map of both at the same time.

I have attached the image and YAML files of our maps.

Could you please help me out with this?
[files and pictures of project.zip](https://github.com/hrnr/m-explore/files/2563389/files.and.pictures.of.project.zip)

hrnr, 2018-11-09:
What version are you using? Are you running from the latest master or from the ROS distribution? The maps you posted are very small (almost none), it will never work for those maps. You need big enough overlapping space for merging to be successful. You can read the attached paper, there are some examples if you want to get the idea what setup you need.

MerryMarry, 2018-11-13:
Hello 
We developed our maps in order to make them overlap, also we reduced the estimation confidence to 0.3 and the code works fine now!
Thank you for your help, I'm closing this issue now.

closed 2018-11-13 by MerryMarry