hrnr/m-explore #26: No frontiers even though many unknown spaces

abylikhsanov, 2019-02-18:
When I launch the explore_lite, it starts settting some goals and explore the area but after a minute or so, no frontiers are left but from the map, the area is large and the robot is surrounded by unknown spaces. Have a look at the screenshot:

[image]

Also, my Lidar is a solid state with FOV of 90 degrees.

hrnr, 2019-02-25:
Hi unfortunately screenshot is not useful for debugging at all as you can imagine.

Please provide a reasonably small bag or map_server map files which I can load to ros.

hrnr, 2019-05-13:
closing for inactivity.

closed 2019-05-13 by hrnr