hrnr/m-explore #28: Could not find nearby clear cell to start search

lingfei93, 2019-04-12:
Hello, 

I have this unique issue where I am running cartographer and both gmapping as my SLAM methods. However, when i run Gmapping, m-explore works fine and it can find frontiers to send navigation goals to my robot. However, when I run Cartographer, I cannot seem to find frontiers with  m-explore. I have checked and the robot is not stuck or anything. Is there anything that I can do from M-explore's end?

Thank you!

Rafcin, 2019-04-25:
was this solved?

Rafcin, 2019-04-25:
I have this same issue. Its not solved in larger areas and it continues to persist

lingfei93, 2019-04-28:
Hi, sorry for this late reply. I found out that actually the issue is that m-explore work on the assumption that the map is published as containing only 3 values, [EMPTY, UNKNOWN, LETHAL OBSTACLE]. So it will try to find a frontier[unknown space] that is next to an empty space. As cartographer maps will not have this situation, I instead just did small hacks in the code to make the code look for frontiers that are near grey areas instead. Hope this helps!

Rafcin, 2019-04-28:
Could you possibly publish this config and or the line that it occurs.

Rafcin, 2019-04-29:
I would really appreciate that 😅

Rafcin, 2019-04-29:
You would be saving my life

Rafcin, 2019-04-29:
One of my thoughts is to pass frontiers from my modified cartographer to explorer by allowing it to subscribe to the array it publishes. 

In the code where it says 
`auto frontiers = search_.searchFrom(pose.position);`
I believe that could be subbed with the marker array instead right?

C++ is not my strong suit.

hrnr, 2019-05-13:
Hi, I haven't tested this package with cartographer. Can you put somewhere an example of a cartographer map?

As I understand, we basically we need to threshold the occupancy probability rather than relying on cell to be marked EMPTY?

c650, 2019-06-13:
Your understanding is correct, @hrnr.

@Rafcin  nice to see we're in this together :)

Rafcin, 2019-06-13:
I believe @c650 that there exists a map server node that fixes this cartographer issue by publishing a new corrected map. I think I linked it under cartographer on the issues page.

c650, 2019-06-13:
@Rafcin Have you gotten that to work?

Rafcin, 2019-06-13:
I know someone used it and it worked, personally I have not tested it, I switched to RTABMAP instead. I would take a look at something like Raffellos Panther robot and maybe use his sim to test it, you just need to switch to m-explore or explore either or the issue persists in both. http://wiki.ros.org/Robots/Panther

I have to prep for graduation but I'll take a look tomorrow morning.

c650, 2019-06-13:
Whatever I do, I am sticking with Turtlebot3.

c650, 2019-06-13:
I have it working! PR Incoming.

lingfei93, 2019-06-18:
Hi, apologies I've somehow missed this entire thread, I did some tweaking in the mean time actually, and found that it helps more to use explore_lite for cartographer if I used the global costmap to search for frontiers instead of using the map to search for frontiers. Hope this somehow helps as well?

c650, 2019-06-18:
How does it help? Mine seems to do ok with just the map.

gakkiHp, 2019-11-19:
Dear @lingfei93,
I want to know that whether m-explore can directly merge maps collected by GMapping algorithm?

blablebli-robots, 2019-12-12:
Hello,

I'm trying to make explore_lite work with cartographer too. I have modified the code as in c650. The issue I am having is that, specially in cluttered areas the frontiers will keep coming back so exploration will take forever, sometimes will finish without notifying, sometimes it seems it will never finish.

Have anyone experienced something similar? Otherwise the exploration seems to be working great, kudos to the developers.

Cheers.

c650, 2019-12-13:
> The issue I am having is that, specially in cluttered areas the frontiers will keep coming back so exploration will take forever, sometimes will finish without notifying, sometimes it seems it will never finish.
@blablebli-robots maybe you should play with the thresholds in my snippet

Rafcin, 2019-12-14:
I think I fixed this issue decades ago. Let me look for my code when I get
home.

blablebli-robots, 2019-12-16:
@c650 , @Rafcin 
Thank you for your feedback, I'll really appreciate any pointers you can provide to make this package work together with cartographer. I also found that in large spaces the exploration ends prematurely, it has a hard time finding frontiers far away from the robot, perhaps also due to cartographer's map format.

Cheers,
Octavio

blablebli-robots, 2019-12-16:
@c650 , @Rafcin

I believe I have found good thresh combo for my setup, I am running more tests right now, but definitely a much better behavior in small cluttered space and large more open space.

Thank you for all your help,
Octavio

Rafcin, 2019-12-27:
Sorry I haven't responded I haven't had time to boot my rover. From what I
remember all this goes back to explorer lite. The issue happens with
position formats. If you look ExLite it takes some standard format while
cartographer publishes a way different format. If I remember correctly
looking up explore lite + cartographer, a chinese student a while back
created his own fix for this.

Other than that, maybe consider RTAB? Its faster and it works really well.

Humburto, 2021-05-15:
If you're still looking for answers, you might want to give this one a try. You can write a node to convert the Cartographer map data to gmapping map data. Just like what @lingfei93 said:
> I found out that actually the issue is that m-explore work on the assumption that the map is published as containing only 3 values, [EMPTY, UNKNOWN, LETHAL OBSTACLE]. So it will try to find a [...]

Here is my code:
``` Python
#!/usr/bin/python3
import rospy
from nav_msgs.msg import OccupancyGrid

# ***************
#   OBSTACLE
M = 75
#   unknown
N = 50
#   free
# ----0-----
#   unknown
# ***************
def callback(cmap: OccupancyGrid):
    data = list(cmap.data)
    for y in range(cmap.info.height):
        for x in range(cmap.info.width):
            i = x + (cmap.info.height - 1 - y) * cmap.info.width
            if data[i] >= M:  
                data[i] = 100
            elif (data[i] >= 0) and (data[i] < N):  # free
                data[i] = 0
            else:  # unknown
                data[i] = -1
    cmap.data = tuple(data)
    pub.publish(cmap)

rospy.init_node('mapc_node', anonymous=True)
sub = rospy.Subscriber('/cmap', OccupancyGrid, callback)
pub = rospy.Publisher('/map', OccupancyGrid, queue_size=20)

rospy.spin()
```

changjunyuan, 2021-09-21:
`#!/usr/bin/python3
import rospy
from nav_msgs.msg import OccupancyGrid

# ***************
#   OBSTACLE
M = 75
#   unknown
N = 50
#   free
# ----0-----
#   unknown
# ***************
def callback(OccupancyGrid):
    data = list(OccupancyGrid.data)
    for y in range(OccupancyGrid.info.height):
        for x in range(OccupancyGrid.info.width):
            i = x + (OccupancyGrid.info.height - 1 - y) * OccupancyGrid.info.width
            if data[i] >= M:  
                data[i] = 100
            elif (data[i] >= 0) and (data[i] < N):  # free
                data[i] = 0
            else:  # unknown
                data[i] = -1
    OccupancyGrid.data = data
    pub.publish(OccupancyGrid)

rospy.init_node('mapc_node', anonymous=True)
sub = rospy.Subscriber('/map', OccupancyGrid, callback)
pub = rospy.Publisher('/cmap', OccupancyGrid, queue_size=20)

rospy.spin()
`

YXGuan, 2022-04-21:
@Humburto 

Hi there,

How do you remapped cartographer's /map topic to /cmap topic?

Thanks,

changjunyuan, 2022-04-21:
Thank for your email, I will check and reply soon.