hrnr/m-explore #35: Merge multiple maps into one map

Chrisch2410, 2019-09-25:
Hi, we have a problem merging three maps from thre turtlebots into one, i have in the tf tree the connection between the maps into the map merge, but when i try to visualize in Rviz, the map merge just take the values of one of the maps, and not a map like it should be. First we have problems to get the transform between the maps, we did what it said in some commits that appears in [https://github.com/hrnr/m-explore/pull/30/commits](url) but, the problem continues. the configuration of map_merge.launch is .

`<launch>`
`<group ns="map_merge">`
  `<node pkg="multirobot_map_merge" type="map_merge" respawn="false" name="map_merge" output="screen">`
   ` <param name="robot_map_topic" value="map"/>`
    `<param name="robot_map_updates_topic" value="map_updates"/>`
    `<param name="robot_namespace" value="robot_"/>`
    `<param name="merged_map_topic" value="/map"/>`
   ` <param name="world_frame" value="/map"/>`
    `<param name="known_init_poses" value="false"/>`
    `<param name="merging_rate" value="4.0"/>`
    `<param name="discovery_rate" value="0.05"/>`
    `<param name="estimation_rate" value="0.5"/>`
   ` <param name="publish_rate" value="1.0"/>`
    `<param name="estimation_confidence" value="1.0"/>`
`  </node>`
`</group>`
`</launch>`

Frames tree it's like this.

[frames (23).pdf](https://github.com/hrnr/m-explore/files/3654440/frames.23.pdf)

the map that should be the merge between other maps

[image]

the map 1 is like this 
[image]

the map 2 is like this.

[image]

the map 3 is like this.
[image]

the map that should be merge is like this ( visualizing the three maps at the same time).
[image]

andresjguevara, 2019-11-06:
Hola Christian. Pudieron resolver el problema? He visto sus videos de ROS Ignite Academy en youtube (#66 y #67) y pareciera ser que si lograron resolverlo pero nosotros estamos teninedo algunos problemas usando los turtlebots3 en un escenario real (no simulado). Seguimos los pasos que ofrecen en el ROSJect de la academia pero el mapa que se genera no es muy accurate. Habra alguna forma de contactarlo directamente? Muchas gracias!

hrnr, 2019-11-10:
If you are using mode without providing initial transformations, then you need to each to be a bit larger a to have enough overlap. Otherwise the transformation between the maps can not be estimated.

closed 2019-11-10 by hrnr