hrnr/m-explore #36: map_merger subscribes two map topics, but only displaying one

gakkiHp, 2019-11-18:
① I use the following **bag(including two robots)** and **a multi-robots-framework** to process and publish two maps, meanwhile I use m-explore to **merge** them online but I just **see three maps with rviz**, **and the merged map is the same as one of original maps**.
② I see some similar issues here 
[https://github.com/hrnr/m-explore/issues/13](url)
but m-explore works well, and I think **the overlap area is big enough**.  It just could not show the merged map with rviz(one of two original maps instead).
③ I also set the estimation confidence to 0.5
**Data set：**
2robots-hospital.bag
**Commands:**
<pre>
roslaunch multirobot_map_merge map_merge.launch
</pre>
**Results：**
robot0：
[image]

robot1：
[image]

the merged map using offline tool g2o_viewer:
[image]

Termiante:
[image]
[image]

**Topic**
<pre>
robot0/map
robot1/map
</pre>

Question：
1. As the figure shown above, the merged map is the same as one of two, I cannot find out the reason.
2. I find that the status in rviz about the merged map is error, but I can still see the graph. I don‘t know whether it matters.
3. I doubt that maybe there is some problems in data format, but the figure above in terminate shows it works well, so maybe not the data format?

hrnr, 2021-01-03:
Unfortunately, I can't access any of the attachments. Could you please upload somewhere the two maps in map_server format?

SYSUHPK, 2021-01-20:
I think I have solved it, thanks a lot.

closed 2021-01-20 by SYSUHPK