hrnr/m-explore #39: Pausing and Restarting Exploration

rmerriam, 2020-02-23:
Working with a version built for Melodic and it goes well.

What I need to do is stop the exploration at certain times even when a goal is incomplete. This when another sensor detects something of interest. After the processing of the sensor data I need to restart the exploration. 

Can I do this? Or would changes be needed?

blablebli-robots, 2020-02-24:
Hello there,

Try calling exploring_timer_.stop() for pausing and exploring_timer_.start() to resume. You may also have to bounce off makePlan(), (e.g. setting a flag) when paused, as there is no need for it to be running during a pause.

craigatproperly, 2020-07-17:
Old thread, but, was dealing with the same issue. Was considering something like adding an action called explore that could be triggered and cancelled (and configurable to not start exploring automatically).  If something like that was added is that a pull request maintainers would consider merging back; or is there some other approach that should be considered.

hrnr, 2021-01-03:
Would adding simple services for start and stop be enough for your usecase?

rmerriam, 2021-01-03:
No longer interested

closed 2021-01-03 by rmerriam

craigatproperly, 2021-01-04:
> Would adding simple services for start and stop be enough for your usecase?

We ended up doing that in a fork; do you want PR for that?

hrnr, 2021-01-05:
yes, that would be welcomed. Thanks.