hrnr/m-explore #40: Issues while exploring large area with many unreachable frontiers.

blablebli-robots, 2020-02-24:
Hello everyone,

I've been having issues whenever I am exploring a large area with many unreachable frontiers (robot won't fit). The issue is it takes forever to realize (if at all) there are no more new frontiers to explore. Sometimes it just keeps rotating in place forever, as if trying to go somewhere but no path is ever found. Otherwise everything works great. Any pointers as to where to look at to avoid this behaviour?

Cheers

artemiialessandrini, 2020-03-31:
@blablebli-robots 
Hi, could increasing **_min_frontier_size_** param do any difference?

artemiialessandrini, 2020-04-02:
Or you can try to tweak goal_distance_bias in your planner?

blablebli-robots, 2020-04-13:
Thank you for your response. I tried both of your suggestions, however the frontier sizes are large enough to be considered, but otherwise unreachable from a path planning point of view (robot wont fit, but laser beams will). Imagine a rectangular room divided by many vertical columns which are separated from each other but  close enough to each other that the robot footprint wont fit through. After exploration, I end up with a map with lots of large enough unreachable frontiers.

So relaxing goal_distance_bias does not help either. 

Thank You.

lukelu0520, 2020-09-16:
I was wondering if you solve the problem. Did you write code to detect available paths, and set a timeout flag to finish the exploration if there is no available path ?

blablebli-robots, 2021-04-06:
@lukelu0520 not really, I still have this issue. As the mapping is progressing it will be difficult, from my point of view, to get rid of these frontiers as the robot may not be completely enclosed by obstacles until the very end of the exploration. We may not discard them all, like with a timer, as there may be some valid frontiers left in the possibly huge list.

ninja-hu, 2022-07-07:
You might want to increase the planner_patience parameter of the global planner. For large maps, if the planner cannot find a valid path within that time threshold, the robot will perform recovery function (in place rotation) and try again.