hrnr/m-explore #41: AKAZE

photopea, 2020-04-05:
Hi Jiří, is it possible to get in touch with you? I would like to ask you something about your work related with the A-KAZE algorithm. Do you have an email address or any other address? :)

hrnr, 2021-01-03:
Hi, I'm no longer very active here, but this is the right channel. Feel free to post your question here.

photopea, 2021-01-03:
Hi, I ended up implementing another method, not AKAZE.  I think you are very talented, is there any chance I could at least partially hire you?

I live in Prague, I studied "MFF UK". Do you live and work in Vienna now?

hrnr, 2021-01-03:
Thanks. You can find my LinkedIn here: https://www.linkedin.com/in/hrnrj/.

I'm currently based in Vienna. Photopea is very cool project, btw!

closed 2021-01-03 by hrnr

photopea, 2021-01-03:
Thanks! I do not use LinkedIn. I even think I would have to pay to be able to write you a message.

Why don't you publish your email address? Is your spam filter really that bad? :D Could you write me an email to support@photopea.com ?