hrnr/m-explore #50: Assertion error when merging with four robots' map

SYSUHPK, 2020-10-11:
Hello, when I try to merge the 2d occupancy grid maps with the ros package you provide(That's a nice work!).
It works wll when I publish two robots' maps. But when I try to merge four robots' data with it, I find it reports an error shown as below:
  
[image]

I have tried to solve it in many ways but failed. It seems that the Assertion error should not have happened for roi.size() must equals to warped_grid.size()
I can't understand why it throws this error. Do you have any idea about it? @hrnr

hrnr, 2020-10-11:
Thanks for the report, looks indeed like a bug.

Could you send me the maps you are trying to merge in map_server format, so
I could reproduce it?

Cheers,
Jiri

SYSUHPK, 2021-01-20:
Sorry to reply so late, I have been dealing with some other things these months. Well, the problem is avoided by skipping some dataset inputs, without fixing it. I need some time to reproduce it.

closed 2021-01-20 by SYSUHPK