hrnr/m-explore #51: uninitialised variables mx, my inside the function FrontierSearch::searchFrom(geometry_msgs::Point position) in frontier_search.cpp

skpro19, 2021-01-05:
I am pasting a code snippet from frontier_search.cpp file. 

`std::vector<Frontier> FrontierSearch::searchFrom(geometry_msgs::Point position)
{
  std::vector<Frontier> frontier_list;

  // Sanity check that robot is inside costmap bounds before searching
  unsigned int mx, my;

  ROS_INFO("Inside the FrontierSearch::searchFrom function.");  

  ROS_INFO("mx: %u my: %u", mx ,my);

  if (!costmap_->worldToMap(position.x, position.y, mx, my)) {
    ROS_ERROR("Robot out of costmap bounds, cannot search for frontiers");
    return frontier_list;
  }`

Here the variables mx and my are uninitialised.  I got the following values for mx and my - `mx: 0 my: 3227058176`. Do these values  make sense or are they garbage values?

skpro19, 2021-01-05:
Also, what does worldToMap(position.x, position.y, mx, my) return? Does it return an ordered pair (x,y)? If true, are we then checking the value of costmap_[x][y] to see if it is 1 or not?

hrnr, 2021-01-10:
We should initialise those variables to comply with: https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#es20-always-initialize-an-object

However, functionally there is nothing wrong with this code as we never use the uninitialised memory.

`worldToMap` returns bool.

skpro19, 2021-01-11:
I have confusion regarding the `worldToMap` function. In the official documentation, the `worldToMap` function takes `mx`, `my` as input. What is the significance of `mx`, `my` variables? 

Also, in `frontier_search.cpp` file, we have a line of code that uses uninitialised `mx` , `my` variables. 

`unsigned int clear, pos = costmap_->getIndex(mx, my)`;

skpro19, 2021-01-11:
Nevermind. I understood it. 
Thanks for your response.

closed 2021-01-11 by skpro19