hrnr/m-explore #52: issue when exploring the large open area

sihanh, 2021-01-07:
when I tried to use explore_lite in a large open area, the robot sometimes stuck in place and never moved again. There are available frontiers that can be visualized viaMarkerArray with rviz but the exlor_lite doesn't send any of them to move_base goal action. Also when I manually set the goal point, the robot can move toward the point but shortly after will be drag back to the stuck location where explore_lite sent. This also happens when robot spawns in a position where facing walls and no frontier explored.

hrnr, 2021-01-10:
Could you share a map where this happens? Ideally, a reproducer in a simulated environment would be very helpful.

sihanh, 2021-01-11:
> Could you share a map where this happens? Ideally, a reproducer in a simulated environment would be very helpful.

added pictures describing the dilemma, the goal can only be sent effectively when frontiers are red

closed 2021-01-19 by sihanh