hrnr/m-explore #55: Ends exploration early when boundary is on static map edge

Combinacijus, 2021-02-26:
explorer_lite seems to ignore frontier when it's on the edge of the static map. It's a problem with mappers like iris_lama_ros which uses dynamic sized static map and makes ir bigger when new obstacles are detected outside current map but in case of exploration with explorer_lite it won't make map bigger because robot turns around before it can be done. It has effect of mapping only the small part of the map. But not all parts of potential frontier are at the edge of the static map so there might be another problem. In image included I drawn red boxes where it should explore but doesn't. At this point robot think it has explored everything and stopped even though some frontiers are still visualized

[image]

blablebli-robots, 2021-03-11:
Are you using the costmap as input? if so you may try using the map itself and see if the frontiers are detected. From the picture it looks that you are using the costmap. Some mapping packages such as cartographer publish a map with a wide range of values instead of the usual ternary valued (free, occupied unknown), I have no experience with iris_lama, so you may want to check that.

Combinacijus, 2021-03-12:
@blablebli-robots you were right using /map and /map_updates solved the problem, thanks!

My final launch file:
```
<launch>
  <node pkg="explore_lite" type="explore" respawn="false" name="explore" output="screen">
    <param name="robot_base_frame" value="base_link"/>

    <!-- Use mapper map-->
    <param name="costmap_topic" value="map"/>
    <param name="costmap_updates_topic" value="map_updates"/>

    <!-- Use move_base map. This does NOT work-->
<!--    <param name="costmap_topic" value="move_base/global_costmap/costmap"/>-->
<!--    <param name="costmap_updates_topic" value="move_base/global_costmap/costmap_updates"/>-->

    <param name="visualize" value="true"/>

    <param name="planner_frequency" value="0.5"/>
    <param name="progress_timeout" value="30.0"/>

    <param name="potential_scale" value="3.0"/>
    <param name="orientation_scale" value="0.0"/>
    <param name="gain_scale" value="1.0"/>
    <param name="transform_tolerance" value="0.3"/>
    <param name="min_frontier_size" value="0.5"/>
  </node>
</launch>
```

[image]

closed 2021-03-12 by Combinacijus