hrnr/m-explore #61: map drifting

hikashi, 2021-06-28:
Hi, 

I am running RRT exploration on the merged map, which initially adjusted using tf transform to the world frame.
Somehow when I start running exploration and expands the map, it somehow shifted instead of fixing at the origin. U can see the cost map is still somehow reflecting the location of the robots but the map is shifted as shown below:
[image]

Is it because of map expanding in the map merge causing the origin to move?
I did try the known init and unknown init but both of them have the same outcome.
Any feedback would be appreciated.

```
<node pkg="multirobot_map_merge" type="map_merge" respawn="false" name="map_merge" output="screen">
      <param name="robot_map_topic" value="map"/>
      <param name="robot_namespace" value="tb3"/>
      <param name="merged_map_topic" value="map"/>
      <param name="world_frame" value="world"/>
      <!-- <param name="world_frame" value="map"/> -->
      <param name="known_init_poses" value="true"/>
      <param name="merging_rate" value="5.0"/>
      <param name="discovery_rate" value="0.1"/>
      <param name="estimation_rate" value="1.0"/>
      <param name="estimation_confidence" value="1.0"/>
  </node> 

<node pkg="tf" type="static_transform_publisher" name="world_to_mergedmap_tf_broadcaster"  args="0.385 0.385 0 0 0 0 world map 100"/>

  <node pkg="tf" type="static_transform_publisher" name="world_to_$(arg first_tb3)_tf_broadcaster"  args="0 0 0 0 0 0 /map /$(arg first_tb3)/map 100"/>
  <node pkg="tf" type="static_transform_publisher" name="world_to_$(arg second_tb3)_tf_broadcaster" args="0 0 0 0 0 0 /map /$(arg second_tb3)/map 100"/>
  <node pkg="tf" type="static_transform_publisher" name="world_to_$(arg third_tb3)_tf_broadcaster"  args="0 0 0 0 0 0 /map /$(arg third_tb3)/map 100"/>
```

Regards,
Billy

sandilyasg, 2023-03-01:
@hikashi Were you able to solve this problem? Please help

hikashi, 2023-03-02:
I am not able to solve this issue after going through the code.
I believe they are using OpenCV to merge both submap without providing a fixed anchor to the coordinate. Hence, there is a need for using a static transform publisher to publish the TF tree relationship between the submap and merged map.
In short, I think it is unable to resolve the mentioned issue. Especially for map expansion and I believe this is the limitation of the merged map using OpenCV. 

I do suggest increasing the initial submap size to be larger for removing auto map expansion for resolving this issue.