hrnr/m-explore #70: Improper merging when robots have different orientations

phoenixrider12, 2022-06-24:
I am using map merging with 2 real robots and my robots have different initial orientations. The launch file for multirobot_map_merge is as follows:
[image]

Merging is like this:
[image]

sandilyasg, 2023-03-01:
@phoenixrider12 were you able to resolve this issue? If so, can you please post what you did. Thanks

phoenixrider12, 2023-03-01:
> @phoenixrider12 were you able to resolve this issue? If so, can you please post what you did. Thanks

I did solve the issue but don't remember what I exactly did. First of all I was not able to do merging with unknown initial positions. I was only able to do it with known initial positions and there also I mentioned the x,y drift between both the robots but didn't mention the yaw difference between them This was my launch file which worked: https://github.com/phoenixrider12/Multi-VOLTA-Exploration-and-Dynamic-Map-Updates/blob/main/map_merge/launch/map_merge.launch

sandilyasg, 2023-03-01:
@phoenixrider12 Thanks for your reply. I am running it with known initial positions. I tried to use the x,y drift between both the robots but somehow it made the merging worse -- I am looking into bugs related to the package itself in the following links:
https://github.com/hrnr/m-explore/issues/15
https://github.com/hrnr/m-explore/issues/12