hrnr/m-explore #79: Merged map is not centered and does not overlap the single maps correctly

mmatteo-hub, 2023-04-07:
I am using the `noetic` version of the package and I am using trying to merge two maps obtained as projection of 3D maps into 2D.
This is the launch file just for the map merge algorithm: [map_merge.txt](https://github.com/hrnr/m-explore/files/11177492/map_merge.txt)
I assume to know the robot poses since I spawn them in that position.
Each robot has its own `/namespace/map` and I publish a rigid transform from this to the `/map` frame to which I center the merged map. This is done via:
  ` <node name="$(arg robot1_ns)_MapAlign" pkg="tf" type="static_transform_publisher" args="$(arg robot_1_x) $(arg robot_1_y) $(arg robot_1_z) $(arg robot_1_Y) $(arg robot_1_P) $(arg robot_1_R) /map /$(arg robot1_ns)/map 50"/>
    <node name="$(arg robot2_ns)_MapAlign" pkg="tf" type="static_transform_publisher" args="$(arg robot_2_x) $(arg robot_2_y) $(arg robot_2_z) $(arg robot_2_Y) $(arg robot_2_P) $(arg robot_2_R) /map /$(arg robot2_ns)/map 50"/>`

where

`<arg name="robot1_ns" value="robot_1"/>
    <arg name="agent_1" value="1"/>
    <arg name="robot_1_x" value="-1.5"/>
    <arg name="robot_1_y" value="-0.5"/>
    <arg name="robot_1_z" value="0.0"/>
    <arg name="robot_1_R" value="0.0"/>
    <arg name="robot_1_P" value="0.0"/>
    <arg name="robot_1_Y" value="0.0"/>

    <arg name="robot2_ns" value="robot_2"/>
    <arg name="agent_2" value="2"/>
    <arg name="robot_2_x" value="1.5"/>
    <arg name="robot_2_y" value="0.5"/>
    <arg name="robot_2_z" value="0.0"/>
    <arg name="robot_2_R" value="0.0"/>
    <arg name="robot_2_P" value="0.0"/>
    <arg name="robot_2_Y" value="3.14"/>`

are the same parameter of the map merge file.

How can I solve?

Pratik-94, 2023-04-17:
I am facing a similar problem. For some reason the from_map node works alright for the example data provided, but does not work for my data. Can the quality of data be an issue? I also tried varying the parameters like estimation rate etc. but facing the same problem consistently...

Pratik-94, 2023-04-17:
@mmatteo-hub can you attach the merged map and the individual maps here?

mmatteo-hub, 2023-12-13:
@Pratik-94  I developed a new map merger on ROS2 starting from a pre-existing code. The new code is able to handle _N_ agents knowing their starting position (so the corresponding transformation) wrt a reference frame.

closed 2023-12-13 by mmatteo-hub